igorantolic/ai-esp32-rotary-encoder #3: Debouncing

pvlvsk, 2018-07-20:
Hello,

i'm using your library with a Heltec esp32 LoRa and a KY-040 on pins 27,19,25,-1

I'm having trouble that the values go almost always 2 steps instead of one one one encoder turn. How to debounce the encoder correctly?

Thank you!

igorantolic, 2018-07-20:
YOU bave version which doesn't jave semisteps so simplificira divide by 2
I bave rotary which needed 2 steps for 1 Count so i Had to slove problem

pvlvsk, 2018-07-21:
Thank you, now it's working. Can I use a second instance of the AiEsp32RotaryEncoder class for multiple rotary encoders?

pvlvsk, 2018-07-22:
I've tried it, but it's not working. Only the last instantiated encoder works... Do you have a tip for me? Thank you!

igorantolic, 2018-07-23:
bacause of the way it works you will need to do some more significant
changes to library
good luck!

igorantolic, 2021-02-25:
In example code now you have options
Just try them all and see which fits the best your rotary encoder module. I had previously set to 2; but recently I bought some different model and now I have to use 4
//#define ROTARY_ENCODER_STEPS 1 
//#define ROTARY_ENCODER_STEPS 2 
#define ROTARY_ENCODER_STEPS 4

closed 2021-02-25 by igorantolic