igorantolic/ai-esp32-rotary-encoder #7: Typo in current code

ttump, 2018-08-22:


igorantolic, 2018-08-22:
Can you please correct and send pull request. Thanks.

telanoc, 2019-03-30:
I think the reset() method should also set the lastReadEncoder0Pos value, otherwise the next time you call encoderChanged() , you read the negative of how much you reset it by.  If you're doing your own calculations based on the delta change, it messes things up.

igorantolic, 2019-04-01:
Will you do the change of code?

telanoc, 2019-04-01:
Sorry, I already switched to a different encoder library (ESP32Encoder by madhephaestus) and don't have your code on my development system any more.  (also, sorry to the OP - my comment was supposed to be a new issue, not a derailment of yours)

igorantolic, 2019-04-01:
I am glad you solved your problem.

igorantolic, 2021-02-25:
V 1.1 working now

closed 2021-02-25 by igorantolic