igorantolic/ai-esp32-rotary-encoder #9: Wiring

Lithimlin, 2019-01-16:
This isn't code related, but I have issues with my wiring probably.
The only values I get are 0 or 1, or -1 or 0. I have a `STEC12E08` encoder with the terminals A, C and B.
The wiring to GND and Vcc is done as shown on page 11 of [this document](https://cdn-reichelt.de/documents/datenblatt/F100/402097STEC12E08.PDF).
Are terminals A and B the same in both this library's and the document's case?
What else could be the cause of this?

Lithimlin, 2019-01-16:
It's probably worth mentioning that if I connect terminals A and C in the document to the A and B pins, I get only the values 0 and 10. 10 on a full step and 0 on a half-step.

Lithimlin, 2019-01-16:
This is my output for 
```
#define ROTARY_ENCODER_A_PIN TERMINAL_A
#define ROTARY_ENCODER_B_PIN TERMINAL_C
```
all while turning the knob clockwise.

```
+Value: 10
-Value: 0
+Value: 10
-Value: 0
+Value: 10
-Value: 0
+Value: 10
-Value: 0
+Value: 10
-Value: 0
+Value: 10
-Value: 0
+Value: 10
Click!
-Value: 1
Click!
Click!
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
Click!
+Value: 2
-Value: 1
+Value: 2
-Value: 1
Click!
+Value: 2
-Value: 1
+Value: 2
-Value: 1
Click!
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
Click!
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
+Value: 2
-Value: 1
Click!
+Value: 2
-Value: 1
+Value: 2
-Value: 1
```
The following is for
```
#define ROTARY_ENCODER_A_PIN TERMINAL_A
#define ROTARY_ENCODER_B_PIN TERMINAL_B
```
again while turning the knob clockwise

```
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
Click!
-Value: 0
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
Click!
-Value: 0
+Value: 1
-Value: 0
+Value: 1
-Value: 0
+Value: 1
```

closed 2019-01-22 by Lithimlin

igorantolic, 2019-01-22:
I am sorry didn't play with this one for whole year.
I hope you can solve this problem
Igor