igorantolic/ai-esp32-rotary-encoder #15: Error compiling Esp32RotaryEncoder.ino for ESP8266

3APEBO, 2020-02-07:
Hello everybody,
Example sketch compiles without any problems for ESP32 boards, but gives the following error for ESP8266 boards (pin definitions are modified appropriately) : 

Arduino: 1.8.10 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Andre\Documents\Arduino\ESP8266_RotaryEncoder_test\ESP8266_RotaryEncoder_test.ino:1:0:

C:\Users\Andre\Documents\Arduino\libraries\ai-esp32-rotary-encoder-master\src/AiEsp32RotaryEncoder.h:31:2: error: 'portMUX_TYPE' does not name a type

  portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;

  ^

Multiple libraries were found for "AiEsp32RotaryEncoder.h"
 Used: C:\Users\Andre\Documents\Arduino\libraries\ai-esp32-rotary-encoder-master
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
===========================================================
Can somebody help?

fgolemo, 2020-03-31:
Got the same issue.

AlbanT, 2020-09-20:
Same issue here :(

igorantolic, 2020-09-20:
You should read the name of the library :))) Perhaps it is ESP32? Kidding... 
Some good people made a contribution for ESP8266, while some other good people improved ESP32.
Would be nice you can contribute making it again compatible for all devices.

AlbanT, 2020-09-21:
@igorantolic I understand but I think this should be mentioned here as the readme states: **Arduino Esp32 / _ESP8266_ MULTIPLE Encoder Library** Maybe update the readme mentioning this issue!?
Unfortunately I wouldn't know where to start to fix this issue so for my ESP8266 I'll look for an alternative but for ESP32 it does seem to work (I uploaded the code and that worked bit didn't do the wiring yet so couldn't really test).

igorantolic, 2020-09-21:
As you can find out seems like change "Encoder library now supports multiple instances of encoders"
improved promised in title, on the expense of compatibility.

https://github.com/igorantolic/ai-esp32-rotary-encoder/commit/48372435a898eb7bda7e1cc6e8875c88b63d1cdd#diff-
8c15260b6e586b781623a9fa7818e635

Might happen the the author of the change probably didn't take into consideration checking compatibility, nor updating the documentation.
To be honest I didn't work on code for so long (typical lack of time issue) so if you would like to contribute community will be thankful.

I said I was kidding regarding the leading remark. I understand your point of view.

igorantolic, 2021-10-03:
Added support for ESP8266

closed 2021-10-03 by igorantolic