igorantolic/ai-esp32-rotary-encoder #19: Dosnt Compile.

Trackhe, 2021-01-29:
```
/var/folders/k_/4s4567950jdf7pxbn5hl51m40000gn/T/arduino_modified_sketch_815518/Esp32RotaryEncoderBasics.ino: In function 'void setup()':
Esp32RotaryEncoderBasics:65:58: error: no matching function for call to 'AiEsp32RotaryEncoder::setup(setup()::<lambda()>)'
  rotaryEncoder.setup([]{rotaryEncoder.readEncoder_ISR();});
                                                          ^
In file included from /var/folders/k_/4s4567950jdf7pxbn5hl51m40000gn/T/arduino_modified_sketch_815518/Esp32RotaryEncoderBasics.ino:1:0:
/Users/trackhe/Documents/Arduino/libraries/ai-esp32-rotary-encoder-master/src/AiEsp32RotaryEncoder.h:68:7: note: candidate: void AiEsp32RotaryEncoder::setup(void (*)(), void (*)())
  void setup(void (*ISR_callback)(void), void (*ISR_button)(void));
       ^
/Users/trackhe/Documents/Arduino/libraries/ai-esp32-rotary-encoder-master/src/AiEsp32RotaryEncoder.h:68:7: note:   candidate expects 2 arguments, 1 provided
exit status 1
no matching function for call to 'AiEsp32RotaryEncoder::setup(setup()::<lambda()>)'
```

igorantolic, 2021-01-29:
>candidate expects 2 arguments, 1 provided

?

Trackhe, 2021-01-30:
Bit this is the example 🤔

closed 2021-01-30 by Trackhe