igorantolic/ai-esp32-rotary-encoder #20: Error compiling for ESP32 - Dev Module

tomsteemson, 2021-02-24:
The initial release compiled just fine, but the new edition glitches on compile with:

no matching function for call to 'AiEsp32RotaryEncoder::setup(setup()::<lambda()>)'

when compiling 

`rotaryEncoder.setup([]{rotaryEncoder.readEncoder_ISR();});`

Issue is in evidence in the Esp32RotaryEncoderBasics.ino example .

This is using the Arduino IDE version  1.8.13 on a Windows 10 Pro platform.

igorantolic, 2021-02-25:
I did some quick fix; but I didn't have time to try it

Can you please get late version now and try
PROVIDE FEEDBACK
thanks

closed 2021-02-25 by igorantolic

tomsteemson, 2021-02-25:
Hi,

It could use a little love; there’s an old version of the code in a separate tab that screws the pooch, but saving the file to a separate folder and deleting the ‘old’ tab and it works beautifully.

Many thanks for the speedy response.

Cheers

Tom Steemson

______________________________________________________________________
This email has been scanned by the Symantec Email Security.cloud service.
For more information please visit http://www.symanteccloud.com
______________________________________________________________________

igorantolic, 2021-02-25:
old file moved to separate folder
:)