igorantolic/ai-esp32-rotary-encoder #22: Guru Meditation Error: when rotating fast

FrankHovis, 2021-04-09:
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400d0b3c  PS      : 0x00060034  A0      : 0x40087284  A1      : 0x3ffbff50
A2      : 0x0000000d  A3      : 0x3ffc1afc  A4      : 0x00000000  A5      : 0x00002000
A6      : 0x00042098  A7      : 0x3ffb8b88  A8      : 0x8008111c  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x8008f099  A13     : 0x3ffb1ad0
A14     : 0x00000000  A15     : 0x3ffb1b0c  SAR     : 0x00000011  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x4008f1b4  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d0b3c

igorantolic, 2021-04-09:
> A14 : 0x00000000 A15 : 0x3ffb1b0c SAR : 0x00000011 EXCCAUSE: 0x00000007nn

IT works on my esp32

stevesch, 2021-04-16:
See this issue-- I think it's the same:
https://github.com/igorantolic/ai-esp32-rotary-encoder/issues/14

don't use the lambda like in the example-- define a separate function with IRAM_ATTR attribute.

igorantolic, 2021-10-03:
Soved. See: https://github.com/igorantolic/ai-esp32-rotary-encoder/issues/14

closed 2021-10-03 by igorantolic