igorantolic/ai-esp32-rotary-encoder #23: not an issue but question: more than one encoder at the same time?

dl8ebd, 2021-04-20:
Hi,
I like your rotary-encoder lib - works great!
Question:
can I use two encoders at the same time for a better menu navigation?

Thanks
Thomas

igorantolic, 2021-04-20:
Regarding "better menu navigation"
Have you ever experienced PRUSA 3D printer menu?
So many options using one encoder.
I strongly believe that is my preferred option rather than multiple
confusing encoders

I had to explain that since that influence an answer first part "can I use
two encoders"
Actually, the code had at some point in time support for that (I am not
sure if it was ever published), and also I am not sure if currently it is
supported. But I saw some forks which do that.

In short, personally, I don't need multiple encoders, and examples like a
complex menu like this for Prusa 3D printer prove that it is so.

I will strongly suggest anyone with a similar question to rethink based on
good examples using a single encoder which might improve CX (Customer
Expirience)

But of course, you can try to find that fork with direct support for 2
encoders.

Cheers!
Igor

dl8ebd, 2021-04-20:
Hello Igor,
thank you very much for your message.
You are right, of course.
I'll rethink the whole thing and try to cope with only one encoder for now.

stay healthy!
Thomas

closed 2021-05-12 by dl8ebd