igorantolic/ai-esp32-rotary-encoder #24: Please give better instructions on connecting bare encoders.

rin67630, 2021-05-02:
Upon connecting bare encoders it is clear that A and B are switched against Vcc (or a pin as pull up).

 But it is  **not obvious** that the button is switched against GND in negative logic.
 Please explain that: I have lost almost two hours to find out, why the button did not work since I did connect it, like A/B on Vcc.

om-hb, 2021-06-12:
For anybody who is finding this issue while searching for an explanation on how to connect a bare encoder.

This should be the way it's done:

[image]

(Original Picture from: https://www.robotics.org.za/EC11-VS-15)

john-, 2021-06-12:
Thanks for posting this.

Based on random references I found I wound up doing things a bit different and things seem to work fine.  Specifically:

C -> 3.3v (not ground)
No pull up (or down) resistors for A/B

Do the pull up/down resistors on A/B add some robustness?

om-hb, 2021-06-12:
> 
> 
> Thanks for posting this.
> 
> Based on random references I found I wound up doing things a bit different and things seem to work fine. Specifically:
> 
> C -> 3.3v (not ground)
> No pull up (or down) resistors for A/B
> 
> Do the pull up/down resistors on A/B add some robustness?

To be honest, I probably don't know more than you do - I also just figured it out by googling. 😅 However, on the rotary encoder modules that you can buy pre-assembled, they seem to use resistors for A/B, so I figured they will probably have a reason to do so. 😄 

[image]

john-, 2021-06-13:
I found another example that includes RC circuit to assist with debouncing:

https://www.electroschematics.com/rotary-encoder-arduino/

Direct link to schematic:

https://www.electroschematics.com/wp-content/uploads/2015/06/rotary-encoder-arduino-wiring.png?resize=305%2C183?w=305

This one does not include wiring of the switch but other examples cover that.

igorantolic, 2021-07-17:
Answered by community

closed 2021-07-17 by igorantolic

rin67630, 2021-07-21:
Just to avoid mistakes: the schematic posted by  om-hb  on 12 Jun does not work with that library.
The contacts a and b must go to 3V Vcc and the (optional) resistors must be pull-down.

john-, 2021-07-21:
> Just to avoid mistakes: the schematic posted by om-hb on 12 Jun **does not work with that library**.

Do you mean this library (ai-esp32-rotary-encoder) ?

> I just have tested this wiring and can confirm that it does not work.
> 
> The contacts a and b must go to 3V Vcc and the (optional) resistors must be pull-down.

Did you mean to say say the contact C must go to VCC?

om-hb, 2021-07-22:
> 
> 
> Just to avoid mistakes: the schematic posted by om-hb on 12 Jun **does not work with that library**.
> I just have tested this wiring and can confirm that it does not work.
> 
> The contacts a and b must go to 3V Vcc and the (optional) resistors must be pull-down.
> 
> For the push switch its just the other logic the contact must go to GND and the input have pull up to Vcc.

I just disassemble my project to double-check: Maybe I'm missing something, I'm confused or there are differences between the encoders, but for me that schematic seems to work just fine!