igorantolic/ai-esp32-rotary-encoder #26: Button is working - encode pin A / pin B not

cmburns, 2021-05-16:
Hi,

I use your library for my encoder (manufacturer ALPS) but I cannot get it working.
Pin A of encoder is connected to IO12, pin B is connected to IO13 and the switch is connected to IO5.
If I run your code I only see in serial monitor if button is pressed - but if I turn the encoder nothing happens.

The default RotaryEncoder-Library (Matthias Hertel) is working. But I want to use yours since it uses interrups.

May you can you tell me what I am missing?

igorantolic, 2021-10-03:
You should be careful about PIN selection. See Andreas Spiess's youtube video on that topic. 
Use pin numbers in example files to verify your encoder works, and then try to select different pins

closed 2021-10-03 by igorantolic

TehseenHasan, 2022-02-03:
> Hi,
> 
> I use your library for my encoder (manufacturer ALPS) but I cannot get it working. I use an ESP32 WROOM. Pin A of encoder is connected to IO12, pin B is connected to IO13 and the switch is connected to IO5. If I run your code I only see in serial monitor if button is pressed - but if I turn the encoder nothing happens.
> 
> The default RotaryEncoder-Library (Matthias Hertel) is working. But I want to use yours since it uses interrups.
> 
> May you can you tell me what I am missing?

I have the same issue ... did you find the solution?