igorantolic/ai-esp32-rotary-encoder #31: Encoder max value is 8192

igorlab, 2021-11-06:
Hi!,  

First of all, thanks for this excellent lib! I try to use it in my project and realized that the max value could be returned by Encoder.readEncoder() is 8192. Is it according to the lib approach or maybe a wrong type of variable is used somewhere?

I try it in Esp32RotaryEncoderBasics example just exclude the string: rotaryEncoder.setBoundaries(0, 1000, circleValues);

Thanks in advance for any prompts!

igorantolic, 2021-11-07:
By skipping custom settings you are back to defaults. (not cool in some situations)

please use 
rotaryEncoder.setBoundaries(-1000000UL, 1000000UL, false);

does it work then?

Another question - in which scenario are you trying to use the example?
Did you go over all examples? (Interesting things you can find in less effort than posting an issue IMHO.

igorantolic, 2021-11-07:
Closing an issue due to misusage. 
Take look at examples and setBoundaries method

closed 2021-11-07 by igorantolic