igorantolic/ai-esp32-rotary-encoder #32: [Question] Which rotary encoders are supported?

eduardomarcos, 2021-11-17:
Hi! 
First of all, thanks for this library, I'm currently using it for my encoder. 
However, I have two kinds of encoders, EC11 and KY-040. The library works good with the KY-040, but with the EC11 is not working properly (just the button click works, the encoder position is flacky). 
More details about he rotary encoders I use:
EC11: https://www.aliexpress.com/item/33039744226.html
KY-040: https://www.aliexpress.com/item/32835918049.html

Regards

igorantolic, 2021-11-17:
seems like missing resistors might cause the issue. If you are not sure how
to fix it by yourself use the other ones. Or send me a few pieces of
problematic ones so I can check

I understand that this solution is not a silver bullet to work with each
and every piece of hardware in this world.

perhaps you wanna play with pulup / pulldown in code or on hardware to fix
the issues

stpavel, 2022-05-15:
> Hi! First of all, thanks for this library, I'm currently using it for my encoder. However, I have two kinds of encoders, EC11 and KY-040. The library works good with the KY-040, but with the EC11 is [...]

EC11 , ESP32 work fine for me. 
I just changed 
`	pinMode(this->encoderAPin, INPUT_PULLDOWN);
	pinMode(this->encoderBPin, INPUT_PULLDOWN);
`

to PULLUP

chcmq, 2022-10-22:
My esp32 development board with E11 also does not work properly, mainly because of the rotation count problem # @stpavel

igorantolic, 2022-10-22:
Someone should send those EC11 to test with them.
"rotation count problem" is way too general information to understand what you are trying to say.
@chcmq - can you try with PULLUP solution mentioned by @stpavel 

We need more info

Bleach665, 2023-04-23:
Confirm that EC11 directly connected (without any resistors, LC contours) to Esp8266 Wemos d1 mini work perfect.

igorantolic, 2023-05-01:
Based on [Bleach665](https://github.com/Bleach665)] response closing the issue

closed 2023-05-01 by igorantolic