igorantolic/ai-esp32-rotary-encoder #34: BugReport

BastelBaus, 2022-02-05:
encoderVccPin is defined as uint, so setting to -1 will not wrok

[AiEsp32RotaryEncoder.h](https://github.com/igorantolic/ai-esp32-rotary-encoder/blob/master/src/AiEsp32RotaryEncoder.h)
Line 50 & Line 71
>> uint8_t encoderVccPin = AIESP32ROTARYENCODER_DEFAULT_VCC_PIN;

Thanks for this nice library :-)

jlfrucot, 2022-02-08:
Maybe, if encoderVccPin is the number of a pin, doesn't it be set negativ ?
Best regards

BastelBaus, 2022-02-08:
I think negative values are meant to be not given. So like VCC pin is not connected to an IO but directly to 3.3V for example. The problem is that all definitions are unsigned, meaning that a -1 is changed to 0xFF = 255 and comparing an unsigned >=0 will always give true. So just all uint_8_t have to be changed to int8_t's and the code will work fine.

jlfrucot, 2022-02-08:
Thanks for this quick and powerful answer !

synack-dev, 2022-10-16:
Since the esp32 has 40 pins (0-39) the pin value check should likely be <40 instead of >=0  as uint8 can never be <0 (a negative number). This will also accomplish a simple sanity check because no pin should be 40 or higher. 
If you prefer I can do a pull request. but I figured since it was such a simple change in one file it can be done manually easier than a pull.

Line 199 
     if (this->encoderVccPin >= 0)
should be
     if (this->encoderVccPin < 40)

Line 207
     if (this->encoderButtonPin >= 0)
Should be 
     if (this->encoderButtonPin < 40)

igorantolic, 2022-10-16:
I prefer goimg bavi to original solution
Signed int and leave the logic as it is and was

That way one should be able ti turn off functionality by setting the vakue
to -1.
At the end od the day that was and still is a simple trick how to reuse a
convenient pin rather than wasting connection to a Vcc 3.3V. Sometimes it
simplify the wiring.

Letting a user set a -1 if thinking that is not the school case sounds
reasonable.

synack-dev, 2022-10-16:
Sure, changing to signed integers will work, however as written would still allow someone to select pins 40-255 which don't exist.
More logic should be put in to limit the pin usage to 0-39. This would be the total proper solution.
Changing the logic fixes both issues at once, but is really more a hack than a fix.

igorantolic, 2022-10-16:
What could happen if someone selects pin 999?
I don't see any problem. I don't think there is a safeguard if I want to
digital read pins 0-999 - it will probably work and no safe guard set. Who
are we protecting and from what?

igorantolic, 2023-05-01:
Resolved returning to int

closed 2023-05-01 by igorantolic