igorantolic/ai-esp32-rotary-encoder #40: For ESP32

VonHirsch, 2022-05-28:
This library is a great idea (boundaries & acceleration) and thank you!  However on ESP32 there are bugs when using the "circleValues" feature (when max go to min and vice versa.)  I would like to try and debug and fix, but when looking at the code, my opinion it needs a whole re-write, and I'm not up to that at the moment unfortunately.

I'm switching to something like this, and adding the other features as needed:
https://github.com/buxtronix/arduino/tree/master/libraries/Rotary

igorantolic, 2022-05-28:
Not šute what was the issue with Circle values but as I remember example
worked

Have you checked all examples using this?

Circle calues should be very simple and usually used on menu selections

In some other use cases that are not covered we should first wrote use
cases and then do the fix to Match the expectations

I understand that many features cause many scenarios so some might not be
tested when using some combinations

VonHirsch, 2022-05-29:
Sure, happy to provide output from the examples.  The only thing I changed in the examples was PIN B to 19 and Encoder Steps to 2 for my circuit and hardware.

```
#define ROTARY_ENCODER_B_PIN 19
#define ROTARY_ENCODER_STEPS 2
```
Here's the output of the "**Esp32RotaryEncoderBasics.ino**" sketch.  Everything working perfectly, Notice I get only one change in value per click of the encoder:

**One Line = One Encoder Click**
[image]

Now here's the output of "**Multi-select.ino**" on the same exact hardware and encoder setup.  I've enabled "Show Timestamp" in the Arduino output and put a red box around those with the same timestamp.  Those are the places where I get more than one change when turning the encoder one click:

**Multiple Changes per Encoder Click (In Red Squares)**
[image]

**Conclusion:** 
When the "Circling" option is enabed there are multiple changes per click of the encoder.

VonHirsch, 2022-05-30:
If I define steps as 4 then the encoder only changes value every other click

igorantolic, 2022-05-30:
ok so for you the best. value is 2

try setAcceleration(0) or (1)
using
rotaryEncoder.setAcceleration(acceletation);

igorantolic, 2022-05-30:
> I thought acceleration might be causing the issue, but when I disable it
> with setAcceleration(0) still the same issue.

I am on a vacation and unable to try this
If you find a solution let me know
It is not expected that if numeric value changes normally that it will
behave differently in this example

VonHirsch, 2022-05-31:
sure no problem.  I looked for an obvious bug in the code but couldn't find it but I hope you can!  Thanks again for the library.

igorantolic, 2023-05-16:
Fixed. (Was a bit complex due to the logic)

closed 2023-05-16 by igorantolic