igorantolic/ai-esp32-rotary-encoder #41: gpio: gpio_set_level(226): GPIO output gpio_num error

ryuenma, 2022-05-31:
Hi, i would like to thank you for providing this library.

I'm currently using it for my project. I'm using wokwi first to code as to grasp what i'm working on without needing to reflash my esp32 many times. everytime i ran the code, i would get this error below:

`E (167) gpio: gpio_set_level(226): GPIO output gpio_num error`

i've asked the wokwi discord, and someone there said (directly quoting them):
`OK, it should be harmless. The docs for the library suggest setting the encoder Vcc pin to -1 when the encoder is connected directly to the ESP's Vcc. But this pin number is stored in an unsigned variable, so when https://github.com/igorantolic/ai-esp32-rotary-encoder/blob/master/src/AiEsp32RotaryEncoder.cpp#L199-L203 that code checks to see if it is >=0 the answer is always true. It's just a warning that the library tried to set pin 255, and there isn't a pin 255.`

Frank-Bemelman, 2022-06-04:
Actually ran into the same warning today. The wokwi comment you found is correct. The encoderVccPin value defaults to -1 when not provided, or you can provide it with -1 yourself when setting up AiEsp32RotaryEncoder(...). 

I can imagine you want to get rid of the warning (like I do too).  You could set a spare pin for encoderVccPin. Or repair the code in AiEsp32RotaryEncoder, line 199.

I changed it to:
	if (this->encoderVccPin < 255)

That made the warning dissapear in my situation.  But as said by the wokwi contributor, it's not a big deal. 

Cheers,
Frank

ryuenma, 2022-06-04:
Alright, thanks frank for the answer!

i'll make do with the workaround for now, as it's nothing fatal.

closed 2022-06-20 by ryuenma