igorantolic/ai-esp32-rotary-encoder #43: Square wave read up and down

julesmoretti, 2022-10-12:
Hello there:

I have a 12-step encoder but I know that one can read typically it at a 24-step rate by being able to differentiate the up and down step of the square wave.

Is there a way to achieve this with this plugin by any chance?

Thank you for your help :)

igorantolic, 2022-10-12:
You can Set how many changes are needed for a single count 1,2 or 4 are
typical values. Ekperimentima Witherspoon these values. In most
implementations we don’t care steps per circle but rather precision of
settings

For large movements use acceleration which will help with that.

sri, 12. lis 2022. u 02:23 Jules Moretti ***@***.***> napisao
je:

igorantolic, 2022-11-14:
Not planned to be implemented here due to lack of hardware.

closed 2022-11-14 by igorantolic