igorantolic/ai-esp32-rotary-encoder #55: Slow void loop...

HoMsEyy, 2023-02-16:
Hi, I have an issue that I can't seem to fix. The main void loop() loops really slow when I use this library. I flashed the Number-select example to my esp-32 dev board and have nothing connected to the board other than my rotary encoder. I also added a serial print to the main loop to check how often it runs, and it prints every ~400ms. However, when I press the Encoder button, the program runs smoothly and loops every 10ms.

This causes problems to my actual program because the main loop is so slow. Is there anything I missed?

igorantolic, 2023-02-17:
you are doing something wrong
recheck examples
recheck button pin has pullup (software or hardware)

HoMsEyy, 2023-02-17:
I honestly don't know what else to do. I have 2 dev boards and 2 rotary switches (KY-040) and they both have this issue. As soon as I disconnect the switch wire (or push on the button), the loop runs at full speed. I'm connect to pins 33 for the switch and 12 and 25 for the clk/dt.

I also tried changing the pins, but same result. Here is my code and a sample of the Serial Monitor:

Th first entries are me holding the switch down, the last 3 are when I release the switch.

08:10:14.123 -> 43484
08:10:14.123 -> 43484
08:10:14.123 -> 43485
08:10:14.451 -> 43816
08:10:14.779 -> 44146
08:10:15.104 -> 44476

`#include "AiEsp32RotaryEncoder.h"
#include "AiEsp32RotaryEncoderNumberSelector.h"
#define ROTARY_ENCODER_A_PIN 4
#define ROTARY_ENCODER_B_PIN 33
#define ROTARY_ENCODER_BUTTON_PIN 17
#define ROTARY_ENCODER_STEPS 4
AiEsp32RotaryEncoder *rotaryEncoder = new AiEsp32RotaryEncoder(ROTARY_ENCODER_A_PIN, ROTARY_ENCODER_B_PIN, ROTARY_ENCODER_BUTTON_PIN, -1, ROTARY_ENCODER_STEPS);
AiEsp32RotaryEncoderNumberSelector numberSelector = AiEsp32RotaryEncoderNumberSelector();

void IRAM_ATTR readEncoderISR()
{
    rotaryEncoder->readEncoder_ISR();
}

void setup()
{
    Serial.begin(115200);
    rotaryEncoder->begin();
    rotaryEncoder->setup(readEncoderISR);
    numberSelector.attachEncoder(rotaryEncoder);
    numberSelector.setRange(6999000.0, 7000000.0, 10, false, 0);
    numberSelector.setValue(6999500.0);
}

void loop()
{
        Serial.println(millis());
    if (rotaryEncoder->encoderChanged())
    {
        Serial.print(numberSelector.getValue());
        Serial.println(" ");
    }

    if (rotaryEncoder->isEncoderButtonClicked())
    {
        Serial.print("Selected value is ");
        Serial.print(numberSelector.getValue(), 1);
        Serial.println(" ***********************");
    }
}`

igorantolic, 2023-02-17:
In case rotaryEncoder->isEncoderButtonClicked()
Does not fura your needs I would stop using the Logic and implement my own
button click Logic which is nonblockable

I am not saying it is the same effort for you but you can benefit doing so.

That includes a custom Logic for evaluateButtonState which should be
nonblocking call with only a several microseconds delay rather than what
you are mentioning

To be honest intention of this library is to solve good enough a rotary
encoder not the button Logic for which you can find much more Advanced
Solutions

Method you mentioned is implemented with more intention to be simple
instead efficient

HoMsEyy, 2023-02-17:
Ok cool thx, I actually started working on a way to implement the button logic. I simply wanted to know if the 400ms delay was normal.

closed 2023-02-17 by HoMsEyy

igorantolic, 2023-02-17:
> <igorantolic/ai-esp32-rotary-encoder/issue/55/issue_event/8551163913@
> github.com>

bool AiEsp32RotaryEncoder::isEncoderButtonClicked(unsigned long
maximumWaitMilliseconds)
PARAMETER is maximumWaitMilliseconds

can you try

https://github.com/igorantolic/ai-esp32-rotary-encoder/blob/master/src/AiEsp32RotaryEncoder.cpp
debounce is 30ms
delay(30); //debounce

Beside that there is "0 ms nonblocking call":
AiEsp32RotaryEncoder::*isEncoderButtonDown*()

HoMsEyy, 2023-03-09:
Inside AiEsp32RotaryEnconder.h
	
       bool isEncoderButtonClicked(unsigned long maximumWaitMilliseconds = 300);

This is it. The 300 ms wait time to check if the button was pressed made the main void loop refresh every 300 ms. I'm not sure why it was written like this but i fixed the issue by changing this value.

igorantolic, 2023-03-10:
This is a patameter so just adjust it to value you want
Seems like your button is constantly pressed or not pulled up
You should implement nonblocking logic for Long press  and here put a value
like 1 ms for detection

The intention was to have simple logic for the most users using it for fast
clicks