igorantolic/ai-esp32-rotary-encoder #56: Compilation errors with examples

NegativCut, 2023-04-17:
Hi, I am trying out this library but a few errors persist during compilation. Everything is left untouched in the examples, including pin assignments.
I am compiling for ESP32 Dev Module.

`Compiling sketch...
In file included from G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:11:
G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.h:53:32: warning: left shift of negative value [-Wshift-negative-value]
  long _minEncoderValue = -1 << 15;
                                ^~

G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp: In member function 'void AiEsp32RotaryEncoder::begin()':
G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:199:26: warning: comparison is always true due to limited range of data type [-Wtype-limits]
  if (this->encoderVccPin >= 0)
      ~~~~~~~~~~~~~~~~~~~~^~~~

G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:207:29: warning: comparison is always true due to limited range of data type [-Wtype-limits]
  if (this->encoderButtonPin >= 0)
      ~~~~~~~~~~~~~~~~~~~~~~~^~~~

G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp: In member function 'ButtonState AiEsp32RotaryEncoder::readButtonState()':
G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:220:14: error: unused variable '_buttonState' [-Werror=unused-variable]
  ButtonState _buttonState = buttonState;
              ^~~~~~~~~~~~

cc1plus.exe: some warnings being treated as errors
Using library Ai_Esp32_Rotary_Encoder at version 1.4 in folder: G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder 
exit status 1
Error compiling for board ESP32 Dev Module.
`

Are these problems with the library, or some upload setting I have?

igorantolic, 2023-04-17:
> G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEnc.h:53:32:

I can see someone changed int to unsigned int vccencoderpin

Until I find some spare time if you know you can modify the code

If you dont need negativne values minencoder can be 0

So with 2 minute change you can fix it
And when I find some time I have to recheck all examples

Some people ask for change and I guess I trusted them more than I should

NegativCut, 2023-04-17:
okay, that cleared up some of it, thanks. Still got the following errors reported at compilation;

G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp: In member function 'void AiEsp32RotaryEncoder::begin()':
G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:207:29: warning: comparison is always true due to limited range of data type [-Wtype-limits]
  if (this->encoderButtonPin >= 0)
      ~~~~~~~~~~~~~~~~~~~~~~~^~~~
G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp: In member function 'ButtonState AiEsp32RotaryEncoder::readButtonState()':
G:\Dropbox\CODE\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:220:14: error: unused variable '_buttonState' [-Werror=unused-variable]
  ButtonState _buttonState = buttonState;
              ^~~~~~~~~~~~

igorantolic, 2023-05-01:
Corrected - returned to be int again as designed initially.
Should work now again

closed 2023-05-01 by igorantolic