igorantolic/ai-esp32-rotary-encoder #58: push-button works on release not press

billbill100, 2023-05-03:
Using the basic sketch,
the push-button does not register as pushed until AFTER it has been released.

I need it to register the push & whilst still pushed, go off and do something....

igorantolic, 2023-05-03:
Use manual logic - your own solution
You can find so many solutions on internet and ChatGPS for Arduino button

Just on each loop check and remember the state of the button (bool
lastStateWasOn, lastStateChangedAt (use millis() function))
Then do the logic

There is absolutely no strings attached - so you can completely ignore the
logic in library

igorantolic, 2023-05-16:
in this version try changing true/false for
    rotaryEncoder.isButtonPulldown = false;
	  rotaryEncoder.areEncoderPinsPulldownforEsp32 = true;