igorantolic/ai-esp32-rotary-encoder #63: Invalid Pin Selected

jbawgs, 2023-05-30:
Hi,

I'm attempting to use your library and get the following error when using -1 for the VCC pin:

[    90][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (66) gpio: gpio_set_level(226): GPIO output gpio_num error

This is in platformio

igor-antolic, 2023-05-31:
Version 1.4 or 1.5?

jbawgs, 2023-05-31:
Whichever version platformio includes in their library options. I changed the lib_dep entry to refer to this GitHub instead (per a similar recent issue) and the problem was resolved.

igor-antolic, 2023-05-31:
I published this last version as 1.5 so should soon be also available on platformio I guess

________________________________
Šalje: jbawgs ***@***.***>
Poslano: Wednesday, May 31, 2023 12:49:05 PM
Prima: igorantolic/ai-esp32-rotary-encoder ***@***.***>
Kopija: Igor Antolic ***@***.***>; Comment ***@***.***>
Predmet: Re: [igorantolic/ai-esp32-rotary-encoder] Invalid Pin Selected (Issue #63)

Whichever version platformio includes in their library options. I changed the lib_dep entry to refer to this GitHub instead (per a similar recent issue) and the problem was resolved.

—

Napomena:
Ova elektronička poruka i njeni prilozi mogu sadržavati povjerljive i/ili povlaštene informacije, a namijenjena je isključivo za upotrebu naznačenog primatelja. Ako ste ovu poruku primili greškom, molimo Vas da o tome bez odlaganja obavijestite pošiljatelja, da uništite izvornu poruku i njene priloge i bez odlaganja poruku i njene priloge izbrišete iz Vašeg sistema. Neovlaštena uporaba, distribucija, otkrivanje, umnožavanje ili izmjena ove poruke je zabranjena. Raiffeisenbank Austria d.d. ne daje niti ne prihvaća pravno obvezujuće izjave putem elektroničkih poruka osim ukoliko drugačije nije izričito navedeno. Budući da komunikacija internetom nije zaštićena, Raiffeisenbank Austria d.d. ne prihvaća odgovornost za sadržaj ove poruke, kao ni za eventualnu štetu nastalu zbog zaraženosti ove poruke virusom ili drugim štetnim programom, te zbog eventualnih tehničkih problema prilikom dostave ove poruke.

Disclaimer:
This e-mail message and any attachment may contain confidential and/or privileged information and is intended for use by the indicated addressee only. If you have received this message in error, please notify the sender immediately, destroy the original message and any attachment and delete this message and any attachment from your system. Unauthorized use, distribution, disclosure, reproduction, or alteration of this e-mail message is forbidden. Raiffeisenbank Austria d.d. neither makes nor accepts legally binding statements via e-mail unless otherwise stated. Considering that internet communication is not secured, Raiffeisenbank Austria d.d. is not responsible for the content of this message, for potential damage occurred due to infection of this message with a virus or other malicious program, and for potential technical problems during delivery of this message.

jbawgs, 2023-05-31:
Thank you, I am enjoying your library

igorantolic, 2023-06-03:
Corrected the library version to 1.6 since 1.5 had the wrong one. Should soon be recognized by the Arduino library crawler,

closed 2023-06-03 by igorantolic