igorantolic/ai-esp32-rotary-encoder #71: Arduino IDE 2.2 Standard example Esp32RotaryEncoderBasics.ino fails to compile with the board TTGO T1.

rin67630, 2023-11-13:
Hi,
I wonder, that a freshly installed Arduino IDE 2.2 fails to compile the standard example Esp32RotaryEncoderBasics.ino with the board TTGO T1.

I get error message

```
d:\Activities\3_Maker\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp: In member function 'ButtonState AiEsp32RotaryEncoder::readButtonState()':
d:\Activities\3_Maker\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:279:14: error: unused variable '_buttonState' [-Werror=unused-variable]
  ButtonState _buttonState = buttonState;
              ^~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors

exit status 1
```

Compilation error: exit status 1

Is the library broken for ESP32 hardware?

Compiling it for ESP8266 gets warnings but finishes gracefully.

AyrtonRicardo, 2024-01-19:
Inside your `File  > Preferences`, select `Compile Warnings = Default`, this problem is caused by your Arduino IDE treating any warning as error.

igorantolic, 2024-03-03:
Fixed in code

closed 2024-03-03 by igorantolic