igorantolic/ai-esp32-rotary-encoder #74: setValue for AiEsp32RotaryEncoderNumberSelector is not working anymore

bardoo, 2024-01-26:
Hi, and thanks for this nice library. I tried upgrading from 1.4 to 1.6, but it seems that the `setValue()` on the AiEsp32RotaryEncoderNumberSelector doesn't do anything anymore. It compiles and everything seems fine, but it just doesn't update with the new value. Let me know if you need more info.

kr4fty, 2024-02-02:
Aren't you referring to the setCount() function?, which sets the encoder counter to the value

void setCount(int64_t value);

DanielPineiroCalvo, 2024-02-15:
Hi! I also I´m looking to modify the encoder counter value, but the funtion setEncoderValue(long newValue) does not compile. In my project the encoder send OSC values, but also can be change by OSC incoming menssages.  Thanks for your help!

DanielPineiroCalvo, 2024-02-15:
Hi again!
 I got it, the function is rotaryEncoder.reset(newValue)

Thanks four you work in this great library!!!

closed 2024-03-03 by igorantolic