igorantolic/ai-esp32-rotary-encoder #75: Compile error

Ratinger59, 2024-02-28:
Hello,
i can't compile the example of the Library, I always get two errors:
----------------------------------------------------------
In file included from C:\Users\xxx\AppData\Local\Temp\.arduinoIDE-unsaved2024128-2216-3fdofb.142c6\Esp32RotaryEncoderTheShortestExample\Esp32RotaryEncoderTheShortestExample.ino:1:
c:\Users\xxx\Documents\Arduino\libraries\Ai_Esp32_Rotary_Encoder\src/AiEsp32RotaryEncoder.h:53:32: warning: left shift of negative value [-Wshift-negative-value]
  long _minEncoderValue = -1 << 15;

and:

In file included from c:\Users\xxx\Documents\Arduino\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:11:
c:\Users\xxx\Documents\Arduino\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.h:53:32: warning: left shift of negative value [-Wshift-negative-value]
  long _minEncoderValue = -1 << 15;
                                ^~
c:\Users\xxx\Documents\Arduino\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp: In member function 'ButtonState AiEsp32RotaryEncoder::readButtonState()':
c:\Users\xxx\Documents\Arduino\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:279:14: error: unused variable '_buttonState' [-Werror=unused-variable]
  ButtonState _buttonState = buttonState;
              ^~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors
---------------------------------------------------------------

Unfortunately, I'm still a beginner and don't know my way around very well, what can I do now?

Thanks
Michael

Ratinger59, 2024-02-28:
> Hello,
> i can't compile the example of the Library, I always get two errors:
>
> In file included from [...]

> In file included from c:\Users\xxx\Documents\Arduino\libraries\Ai_Esp32_Rotary_Encoder\src\AiEsp32RotaryEncoder.cpp:11: [...]

kr4fty, 2024-02-29:
It is an error in the compiler and not in the library. 

Read this thread for more detail:
[ESP32 unused variable compile error](https://forum.arduino.cc/t/esp32-unused-variable-compile-error/1039022/4)

Ratinger59, 2024-03-02:
Ok, thanks for this Information. I have changes the txt file of Arduino IDE and i become only warnings.
But... why i can't find the declaration of the variable in AiEsp32RotaryEncoder.cpp, line 279: _buttonState? (first char is underline!)

ButtonState AiEsp32RotaryEncoder::readButtonState()
{
	ButtonState **_buttonState** = buttonState;
	// buttonState =
	return buttonState;
} 

and the point of AiEsp32RotaryEncoder.h line 53 i have changed -1 to 0xFFFFFFFF, then the IDE works fine. ;-)

best regards
Michael

igorantolic, 2024-03-03:
Fixed

closed 2024-03-03 by igorantolic

igorantolic, 2024-03-03:
Ha ha
You should be able to find it since it is in text you provided
ButtonState *_buttonState* = buttonState;

This was just never used (probably some logic was removed but variable
remains)

So compiler gave us a warning
And when you treat warning as error it become an error which it actually
isnt.