igorantolic/ai-esp32-rotary-encoder #76: Error when using with WT32-ETH01 Board

rybitski, 2024-03-12:
When compiling on the WT32-ETH01 (ESP32 based board with wired ethernet) The board spits out an error in the serial monitor on boot
`E (9) gpio: gpio_set_level(226): GPIO output gpio_num error`

rybitski, 2024-03-12:
Certainly, changing the pins is how I was able to make this function for the WT32-ETH01. I've loaded the "Esp32RotaryEncoderBasics" example.
Below are the pin assignments I've chosen. 
`#if defined(ESP8266)
#define ROTARY_ENCODER_A_PIN D6
#define ROTARY_ENCODER_B_PIN D5
#define ROTARY_ENCODER_BUTTON_PIN D7
#else
#define ROTARY_ENCODER_A_PIN 39
#define ROTARY_ENCODER_B_PIN 36
#define ROTARY_ENCODER_BUTTON_PIN 15
#endif
#define ROTARY_ENCODER_VCC_PIN -1 /* 27 put -1 of Rotary encoder Vcc is connected directly to 3,3V; else you can use declared output pin for powering rotary encoder */
`
Regardless of what I change these values to I still get the error. To be clear- the code complies, but when the ESP32 reboots in spits out the error on the serial monitor. Somewhere in the library (or included) there is a reference to a pin that is not listed here. 
For reference here is some info on the dev board I'm experiencing the issue with: [link](https://github.com/ldijkman/WT32-ETH01-LAN-8720-RJ45-?tab=readme-ov-file)

igorantolic, 2024-03-12:
and you verified documentation?
https://www.esp32.com/viewtopic.php?t=33952
or youtube Spiess Andreas pins I don't use for ESP32

rybitski, 2024-03-13:
I've used this board in multiple projects up to this point. I can't say for certain that the documentation is 100% correct. 
I can say that if I connect switches to pins 36, 39, and 15 that I can use those as inputs without issue. The problem only arises when compiling and running this library.  (which is quite nice- hence why I'm reporting an issue to try and contribute)
I know it is tempting the think commenters are overlooking simple things, but this appears to be a legitimate issue.
I've tried the following  assignments with the same results. 
```#define ROTARY_ENCODER_A_PIN 14
#define ROTARY_ENCODER_B_PIN 12
#define ROTARY_ENCODER_BUTTON_PIN 15```
```#define ROTARY_ENCODER_A_PIN 2
#define ROTARY_ENCODER_B_PIN 4
#define ROTARY_ENCODER_BUTTON_PIN 12```
Thanks for your responses. It can be hard to support ALL ESP32 hardware especially if you don't own it. I'm happy to look into sending you a dev board if you are interested in trying to reproduce.

rybitski, 2024-03-13:
UPDATE-
If I assign a pin to VCC PIN instead of -1 the issue goes away.
```
#define ROTARY_ENCODER_VCC_PIN 4
```
Looks like there is an issue with passing `-1` as a pin.

rybitski, 2024-03-13:
AiEsp32RotaryEncoder.cpp line 135
`AiEsp32RotaryEncoder::AiEsp32RotaryEncoder(uint8_t encoder_APin, uint8_t encoder_BPin, uint8_t encoder_ButtonPin, uint8_t encoder_VccPin, uint8_t encoderSteps)`

-1 does not fit in uint8_t

igorantolic, 2024-03-13:
use the last version. It is int not unit anymore

see

https://github.com/igorantolic/ai-esp32-rotary-encoder/issues/63

and
https://github.com/igorantolic/ai-esp32-rotary-encoder/blob/master/src/AiEsp32RotaryEncoder.cpp#L193

AiEsp32RotaryEncoder::AiEsp32RotaryEncoder(uint8_t encoder_APin, uint8_t encoder_BPin, int encoder_ButtonPin, int encoder_VccPin, uint8_t encoderSteps, bool areEncoderPinsPulldown_forEsp32)

rybitski, 2024-03-13:
I will install the library manually- it looks like my Arduino 2.3.2 IDE thinks 1.4 is the latest. Sorry to open up an old issue.
[image]

igorantolic, 2024-03-14:
for me   Arduino 2.3.2 IDE says 1.6
Ininstall and then verify if there is a newer version on dropdown left of
the install button
Update seems to be moved (hidden) on  Arduino 2.3.2 IDE - 3 dots menu on
the right / Other versions

Please verify that

closed 2024-06-21 by igorantolic