igorantolic/ai-esp32-rotary-encoder #78: rotaryEncoder.setEncoderValue() offset by one when value is negative

jasontaubman, 2024-04-03:
Hello,

When using rotaryEncoder.setEncoderValue() to set a negative value, and then reading the resulting encoder value with rotaryEncoder.readEncoder() the result is always off by one towards zero. For example:

```
void setup() {
  Serial.begin(115200);
  rotaryEncoder.begin();
  rotaryEncoder.setup(readEncoderISR);
  bool circleValues = false;
  rotaryEncoder.setBoundaries(-1000, 1000, circleValues); 
  rotaryEncoder.setAcceleration(250);  
  
 test = -10;
  
  rotaryEncoder.setEncoderValue(test);

}

void loop() {

  Serial.println(rotaryEncoder.readEncoder());

}
```

This code results in printing the value -9 when -10 would be expected. 

Setting the encoder to a value >= 0 works as expected.

kr4fty, 2024-04-12:
setEncoderValue calls the reset() function, inside AiEsp32RotaryEncoder.cpp

```
void AiEsp32RotaryEncoder::setEncoderValue(long newValue)
{
	reset(newValue);
}
```

and within the reset() function there is a "correction" parameter that is testing the displacement by 1, in this case, of the set value

```
void AiEsp32RotaryEncoder::reset(long newValue_)
{
	newValue_ = newValue_ * this->encoderSteps;
	this->encoder0Pos = newValue_ + this->correctionOffset;
	this->lastReadEncoder0Pos = this->encoder0Pos;
	if (this->encoder0Pos > this->_maxEncoderValue)
		this->encoder0Pos = this->_circleValues ? this->_minEncoderValue : this->_maxEncoderValue;
	if (this->encoder0Pos < this->_minEncoderValue)
		this->encoder0Pos = this->_circleValues ? this->_maxEncoderValue : this->_minEncoderValue;

	this->lastReadEncoder0Pos = this->readEncoder();
}
```

If you eliminate that value correction, it would be showing you the real value

```
- this->encoder0Pos = newValue_ + this->correctionOffset;
+ this->encoder0Pos = newValue_; //+ this->correctionOffset;
```

Steve8291, 2025-01-01:
I was having a very similar issue that I think my be related. My values are always returning at an offset of 1.
Using the [Number-select.ino](https://github.com/igorantolic/ai-esp32-rotary-encoder/blob/master/examples/Number-select/Number-select.ino) with the following values will not work:
```
numberSelector.setRange(31, 159, 32, false, 0);    // minValue, maxValue, stepIncrement, cycleValues, decimalPlaces
numberSelector.setValue(127);
```
The set values that I print range from (0, 32, 54, 96, 128, 160)
Setting even values like below works correctly but causes me to have to add and subtract for use in my program:
```
numberSelector.setRange(32, 160, 32, false, 0);    // minValue, maxValue, stepIncrement, cycleValues, decimalPlaces
numberSelector.setValue(128);
```