igorantolic/ai-esp32-rotary-encoder #79: click extension

OpenSourceDevelop, 2024-05-11:
Thanks for the practical library, I really enjoy using it. But what would be very practical if it had integrated click, double click and long click detection. Maybe a suggestion 😋 Thank You!

john-, 2024-05-11:
As I recall I just use my own logic for the button and this library only for the rotary encoder. It work pretty well and doesn't cause any problems.

igorantolic, 2024-05-11:
I think the this lib should not provide this
Since the described but much more like long press, longer press and extralong press should all go with some of suggested and make a new button utility library

Such a library should be separated razher than incorporated in this one.

Because button utility is more generic than ONLY a rotary button states

Anyone should be able to define a lot if buttons in application each having its own parameters like what is considered as long ot extra long press

I use such functionalities to enter some special menus and even limit the user that special menu enter can be done within 30 seconds after powering the device

So I am totally for creating a sepatate and smart library for multiple button handling.

igorantolic, 2024-05-11:
Please anyone else guve us your thoughts vefore closing this as out of scope.

In case someone use some of similar existing libraries feel free to share exipirience here. Probably someone in his lib handles this with much more details than we shoud here.

igorantolic, 2024-06-21:
Closed since considered out of scope of this lib (at least for now).
Anyway thanks for the idea, and example that people can use

closed 2024-06-21 by igorantolic