igorantolic/ai-esp32-rotary-encoder #85: Encoder Change is Inverted

WarMacre, 2024-11-02:
On the encoder that I am using (Chinese KY-040 Module), turning clockwise reduces the changed amount and visa-versa. Is there a way to switch this? If not, can I request an addition to the next build to invert the change amount?

Steve8291, 2025-01-01:
The CLK and DT pins are reversed on that module. You have two options. Do one or the other but not both:
1) Reverse the A and B pin variables you pass when make an instance of the enocder like
```
AiEsp32RotaryEncoder rotaryEncoder = AiEsp32RotaryEncoder(ROTARY_ENCODER_B_PIN, ROTARY_ENCODER_A_PIN, ROTARY_ENCODER_BUTTON_PIN, ROTARY_ENCODER_VCC_PIN, ROTARY_ENCODER_STEPS);`
```
2) Reverse the A and B pin defines:
If you had
```
#define ROTARY_ENCODER_A_PIN 32
#define ROTARY_ENCODER_B_PIN 21
```
Change it to
```
#define ROTARY_ENCODER_B_PIN 32
#define ROTARY_ENCODER_A_PIN 21
```

igorantolic, 2025-01-01:
Closing since not code issue but rather adjusting for different HW as answered

closed 2025-01-01 by igorantolic