igorantolic/ai-esp32-rotary-encoder #87: Any plans to use hardware pulse counting (PCNT) instead of interrupts?

pmi123, 2025-05-21:
Just what the title says, any plans to expand this project to use PCNT instead of interrupts?

igorantolic, 2025-05-22:
Soula YOU Like to explain reasons and advantage and/or limitations

IT is always the question why woukd someone want to change something it
just works

sri, 21. svi 2025. u 20:12 Mark Phillips ***@***.***> napisao
je:

igorantolic, 2025-05-22:
Grmini says IT is not the same and YOU can not  just Count since there
shield be Logic reading 2 signals and understanding direction of movement
not just counting 1 signal that YOU have for example when counting how
motor turns

So I dont recognize it is even possible for this library.

Of course YOU can fork and show how IT works  but also describe advantages

In embedded systems, especially in devices like ESP32, the Pulse Counter
(PCNT) module and interrupts are both used to detect and respond to events,
but they serve different purposes. PCNT is primarily used to count the
number of pulses or edges on a signal, while interrupts are used to alert
the processor to

pmi123, 2025-05-22:
I understand the meaning of "if it ain't broke, don't fix it". I am working on a project that uses an e-ink display and rotary encoder for the user interface.

A google search on [PCNT Advantages over interrupts](https://www.google.com/search?q=esp32+pcnt+advantages+over+interrupts&client=ubuntu-sn&hs=BAJ&sca_esv=2af23fbe5c927cf8&channel=fs&ei=FKUvaLz3KonHkPIPotrY6Qg&ved=0ahUKEwi80byFkLiNAxWJI0QIHSItNo0Q4dUDCBA&uact=5&oq=esp32+pcnt+advantages+over+interrupts&gs_lp=Egxnd3Mtd2l6LXNlcnAiJWVzcDMyIHBjbnQgYWR2YW50YWdlcyBvdmVyIGludGVycnVwdHMyCBAAGIAEGKIEMggQABiABBiiBDIFEAAY7wUyBRAAGO8FMggQABiABBiiBEjiKlC2CliiKHADeACQAQCYAXKgAZINqgEDOC45uAEDyAEA-AEBmAISoALJDMICChAAGLADGNYEGEfCAggQIRigARjDBMICChAhGKABGMMEGAqYAwCIBgGQBgiSBwQ3LjExoAfmK7IHBDQuMTG4B7wMwgcGMC41LjEzyAdE&sclient=gws-wiz-serp) yields the following:

> The ESP32 PCNT (Pulse Counter) offers several advantages over using GPIO interrupts for counting pulses, particularly in applications requiring high precision and efficiency. PCNT provides dedicated hardware for pulse counting, allowing for accurate pulse detection, glitch filtering, and watchpoint interrupts without relying on software-based interrupt handlers. This results in faster and more deterministic counting, reduced CPU overhead, and the potential for more complex counting scenarios like quadrature decoding. 

I have a [KY-040 rotary encoder](https://joy-it.net/en/products/COM-KY040RE) hooked up to an ESP32 S3 DevKitC-1 using the [RotartyEncoderPCNT library](https://github.com/vickash/RotaryEncoderPCNT/blob/main/README.md). There are many other libraries for rotary encoders, this one was the simplest. The position data returns increasing values ( -2,-1,0,1,2,3,4....) when the knob is rotated clockwise the values are decreasing (4,3,2,1,0,-1,-2) when rotated counter-clockwise, so direction (`last_position - current_position > 0`) and position are available. This library uses both outputs from the rotary encoder. 

I am interested in using PCNT with your code for the reasons listed above as well as a smaller code base for my project. Granted, not all ESP32's have PCNT hardware, but for those that do have it (an increasing number, I believe) it would add a new feature to your code base.

igorantolic, 2025-05-23:
That is nice
Looking forward to see it in action

I guess bi smaller code base you mean binary size difference although for
esp32 if is rarely important compared to small memory like Arduino uno.

Keep us updated about your solution.

Cheers

pet, 23. svi 2025. u 00:48 Mark Phillips ***@***.***> napisao
je:

closed 2026-01-04 by igorantolic